MarlinFirmware/MarlinDev #411: Broken SHA hash on MarlinDev-MarlinPlatform1.6.7b.zip

thinkyhead, 2016-04-24:
I updated the `MarlinDev-MarlinPlatform1.6.7b.zip` download to include the latest Sanguino `pins_arduino.h` and fixed some paths. After I updated the branch I downloaded `MarlinDev-MarlinPlatform1.6.7b.zip` and computed a SHA-256 hash on it. I placed the SHA-256 hash into the `.json` file and updated the version to `0.0.3`. Arduino IDE complains that the signature is wrong.

Can someone help me fix the SHA-256 hash and show me what I did wrong?

If this is documented someplace, just post a link and I will check it out! Time is short, or I would take more time to figure this out myself…

jbrazio, 2016-04-28:
I've just installed v0.0.3 on IDE 1.6.8 Linux without issues.

esenapaj, 2016-04-29:
I also tried to install ```0.0.3```, but it causes CRC error.

 environment:
Windows 10 Pro Insider Preview Build 14332
Arduino IDE 1.6.9 Hourly Build 2016/04/28 04:33
JRE 9 Build 115
[image]

jbrazio, 2016-04-29:
Is this CRC error the same one @thinkyhead was complaining ?

thinkyhead, 2016-05-02:
CRC error is probably the same thing (checksum doesn't match). Still looking for the correct method to generate the hash. Apparently MD5 is supported too, so I might go with MD5.

bobc, 2016-05-02:
Arduino caches the zip file, and doesn't appear to refresh them so it is easy to get out of sync. Uninstalling the package does not delete the files, so I delete them manually to ensure the latest versions are downloaded.
On windows, zip files are cached in C:\Users\_user_\AppData\Local\Arduino15\staging\packages

esenapaj, 2016-05-02:
@bobc 
I'm thankful to you,
I've been able to install  ```0.0.3``` successfully after deleting all caches in ```%USERPROFILE%\AppData\Local\Arduino15\staging\packages```.

@thinkyhead 
Now CRC error is gone, but new problem become obvious.
I can't install ```0.0.1``` after deleteing cashes.
Because https://github.com/Wackerbarth/MarlinDev/archive/MarlinPlatform-1.6.7-a.zip is no longer availablein in internet.

jbrazio, 2016-05-02:
@thinkyhead sha256sum should work..

thinkyhead, 2016-05-02:
> Because https://github.com/Wackerbarth/MarlinDev/archive/MarlinPlatform-1.6.7-a.zip is no longer available on the internet.

Whoops. For some reason I thought I updated the path to point to our own copy of this. I'll figure out the current path for this and update it soon.

thinkyhead, 2016-05-02:
> Arduino caches the zip file

That's a bug. Obviously it should download the latest version. At the very least it should compare sizes and download one with a different size. I'll post an issue about it on their repo.

jbrazio, 2016-05-02:
@thinkyhead wont we stop supporting this method of distributing Marlin ?

thinkyhead, 2016-05-02:
@jbrazio I know we're dropping the library-based version of Marlin. However, this hardware support stuff might still be useful (and Sanguino and Gen7 apparently need it…?).

jbrazio, 2016-05-03:
Scott I'm really asking, because I don't know what "else" is depending on this "Arduino lib" method. Why do you say Sanguino and Gen7 are dependent on this method ?

thinkyhead, 2016-05-03:
There are two separate things going on. We have the version of Marlin which can be downloaded and installed as a library, as with any other code library. (Arduino IDE > Sketch > Include Library > Manage Libraries…) And then we have Marlin "platform" hardware support, which installs additional "boards" in the Arduino "Board" menu (Arduino IDE > Tools > Board > Boards Manager…). The boards/platform support allows you to build Marlin on boards which are not included by default with Arduino. The Marlin library option allows you to build Marlin based on the MarlinDev layout, where just the `.ino` and configuration files live in a folder by themselves.

closed 2016-05-25 by boelle