MarlinFirmware/MarlinDev #412: Z-probe not raising after home

Tyler-Darby, 2016-04-25:
I've just set up auto leveling for my printer, and it seems that when homing the Z axis, after homing, when the arm tries to retract itself it is not taking into account the Z_RAISE_BEFORE_HOMING number that i've set in my config. I realize that it says "Before homing" but logically shouldn't it take effect after homing as well?

My issue is that after homing when the arm tries to retract itself, my probe hits the bed and cannot properly raise up, causing the probe to get stuck.

jbrazio, 2016-04-25:
No, that option is intended to avoid any possible obstructions you may have on the bed *before* homing. To control ABL you have three options:

```
#define Z_RAISE_BEFORE_PROBING 15   // How much the Z axis will be raised before traveling to the first probing point.
  #define Z_RAISE_BETWEEN_PROBINGS 5  // How much the Z axis will be raised when traveling from between next probing points.
  #define Z_RAISE_AFTER_PROBING 15    // How much the Z axis will be raised after the last probing point.
```
But his only works with ABL's `G29` not with `G28`.

Tyler-Darby, 2016-04-25:
Okay, goof on my part. I was using an old version of Marlin. I am having some other issues though, but I will create a new issue for those since it seems to be unrelated.

closed 2016-04-25 by Tyler-Darby