MarlinFirmware/MarlinDev #413: G29 homes before extending probe

Tyler-Darby, 2016-04-25:
When I execute the G29 command, it seems that it homes before letting down my probe, and then attempts to take measurements.

This is an issue because my Z-axis has to be higher than my home position before it can extend the arm. So, when I execute G29, my Z-axis homes, the arm is let down at the home position (causing it to hit the bed) and then a measurement is attempted (even though the Z-probe isn't actually down).

Here is my marlin config:
[Configuration](https://gist.github.com/Tyler-Darby/09b749e3f43348aa77308d7657083426)

jbrazio, 2016-04-25:
The sequence is `G28` and then a `G29`, is your `G28` correctly homing Z ?

Tyler-Darby, 2016-04-25:
Yes, G28 properly goes to the correct height, extends the probe, homes, and retracts.

G29 goes to the home position that was acquired by G28 and attempts to extend the probe at that position, causing the probe to hit the glass.

Sniffle, 2016-04-25:
You need to adjust your before durung and after heights in the bed leveling
section of the firmware.

Sorry im on my phone typing 1 handed or i would give you more precise names.

Unless it has moved very recently it should be a little below the Z offset
parameter.

thinkyhead, 2016-04-25:
It would also help to know what version of Marlin you are using, in case it's an older one. We aren't patching older ones, but we do want to know if a bug that exists in an older Marlin still exists in the newest code so we can fix it. If you see this problem with `RC6` or `RCBugFix` (and it's not a configuration problem) then we should figure it out soon!

Tyler-Darby, 2016-04-25:
I actually just realized I have been making my issues under the dev repo. 

I'm on either the latest release or the dev version, i'll have to let you know tomorrow. I cloned the repo directly, though, so it's the latest of one of the two.

Tyler-Darby, 2016-04-25:
@Sniffle I've actually already messed with those numbers and they seem to have no effect on this particular behavior.

thinkyhead, 2016-04-25:
@Tyler-Darby Definitely download and test the latest code. https://github.com/MarlinFirmware/Marlin/tree/RCBugFix

gddeen, 2016-04-25:
Mine does the same...

I just always make a point of getting up and going to the printer and raising the Z before a G29 is ever
issued. (too lazy to fix it.)  I modified the values to be higher but it didn't fix it.

I'm using the dev level from 4/20/16. I had the same problem on RCBugFix.

gddeen, 2016-04-25:
It is kinda neat watching the printer smash and destroy the Zprobe. (Mine is for a JHead V5 so the retracting arm is pretty long.) I corrected this by making the probe length just a few mm longer rather than 5mm. But, this means I won't be able to use longer nozzles.

Tyler-Darby, 2016-04-25:
Well, seems that I have solved this problem. 
I had to essentially go trial and error with these 3 values, eventually coming up with some pretty weird numbers for them, but hey, it works. 

`#define Z_RAISE_BEFORE_PROBING 
  #define Z_RAISE_BETWEEN_PROBINGS 
  #define Z_RAISE_AFTER_PROBING `

thinkyhead, 2016-04-25:
@Tyler-Darby On `RCBugFix`? Are you sure you specified all the right probe options? The latest Marlin has added a few new ones. I will look again at your configuration. Please post your newest one if it differs other than the options you just pasted (without their values?).

Blue-Marlin, 2016-04-25:
@Tyler-Darby 
If i understood what you are telling us.
You have z-min endstop for homing.
You have a probe for G29.
You have defened Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

Where is your z-min endstop connected?
Where is your probe connected?
Are they connected in parallel?

Tyler-Darby, 2016-04-26:
@Blue-Marlin My Z-min endstop is my probe. They are connected to Z-Min on the RAMPS board.

@thinkyhead The only values that have changed are:
```c++
#define Z_RAISE_BEFORE_PROBING 25   // How much the Z axis will be raised before traveling to the first probing point.
  #define Z_RAISE_BETWEEN_PROBINGS 5  // How much the Z axis will be raised when traveling from between next probing points.
  #define Z_RAISE_AFTER_PROBING 10  
```
The reason I say that these are weird is because the raise before and after aren't the same. I don't know if they are supposed to be, but since that's how they were originally defined in the config I assumed they probably should stay the same. Either way, it seems to all be working just fine now.

thinkyhead, 2016-04-26:
@Tyler-Darby Your `Configuration.h` (linked in your first post) is incompatible with `RCBugFix` and `RC6`. What config are you using when you build and test `RCBugFix`? (We're only fixing bugs in `RCBugFix`.)

Tyler-Darby, 2016-04-26:
@thinkyhead Oh, my bad, I forgot that the original config I linked is for a different version. I've already closed up shop for the night so I will have to get that for you tomorrow but the values essentially all the same (aside from the three that I mentioned above) just in the new config file.

thinkyhead, 2016-04-26:
Thanks for the update and kudos. There's a lot of love for Marlin and RepRap around here!

With the newest configs and `RCBugFix` it might be easy to miss (or mis-configure) the new probe options. So be sure to check through them and make sure you have all the right "boxes ticked." Since you have a servo probe which is your Z endstop, it shouldn't be too hard to get the options set. But I understand if you find them a little daunting, so don't hesitate to post any questions you have.

gddeen, 2016-04-26:
I'm actually using 110_RC5... I'll switch back to RCBugFix, since when I do 'T1' in RC5 (as of 4/20) it messed up the Z value. IE. Type in T1, Z 0 goes to Z -?%7 on the LCD.

In my settings below, I allowed for 12mm clearance for the probe.

#define X_PROBE_OFFSET_FROM_EXTRUDER -25   // 10  // X offset: -left  [of the nozzle] +right
  #define Y_PROBE_OFFSET_FROM_EXTRUDER -5    // 10  // Y offset: -front [of the nozzle] +behind
  #define Z_PROBE_OFFSET_FROM_EXTRUDER -12.4 // 10  // Z offset: -below [the nozzle] (always negative!)

  #define Z_RAISE_BEFORE_PROBING 25 // 15   // How much the Z axis will be raised before traveling 
  #define Z_RAISE_BETWEEN_PROBINGS 7 // 5  // How much the Z axis will be raised when traveling
  #define Z_RAISE_AFTER_PROBING 5 // 15    // How much the Z axis will be raised after the last pr

gddeen, 2016-04-26:
OK, I grabbed RC6 Today... The 'T1' command which changed the Z axis value is fixed. However.

G28 does not raise the Z axis prior to extending the probe.  If I put the nozzle on the print bed at [100,100,0], then type G28
It goes X0, then to Y0, then moves to the middle, extends the probe (which does not extend because
it hits the bed.) I always start the G28 when the Z height is at or above 20mm.

G29 works well because it DOES raise the Z axis, then extend the Z-probe arm.

gddeen, 2016-04-26:
Ahhh... MIN_Z_HEIGHT_BEFORE_HOMING wasn't set... Thanks!

jbrazio, 2016-04-27:
If this issue is solved for everyone experiencing it, could you please close it ? Thanks.

closed 2016-04-29 by Tyler-Darby

thinkyhead, 2016-05-09:
Someday, perhaps we will add a `SERVO_CLEARANCE` setting for X, Y, and Z servos, so they will all be able to move, as needed, to give space for the arm to rotate.