MarlinFirmware/MarlinDev #414: Stepper motors don't move and the system hangs after define the mix factors

DIYlover, 2016-04-26:
I ran the following commands in pronterface to try extrusion of mix colors but failed.  The system is  hangs after I trigger the extrude button in pronterface.

<Here are the commands>
Extruding 20.000000mm of filament. <--- Here I try to trigger extrude button
Warning: Mix factors must add up to 1.0. Scaling. 
>>>m163 s0 p0.5
SENDING:M163 S0 P0.5
>>>m163 s1 p0.5
SENDING:M163 S1 P0.5
Extruding 20.000000mm of filament.  <--- once I click extrude again, the system is hangs, the LED on mega256 board is keep flashing endless

<Background>
My 3D printer is delta type.  it has 3 extruder motors attached.  2 of them are attached to main board which is mega2560, the last one is attached to external CNC extension board.  The hardware circuit was well tested with old non-multi-extruder support marlin firmware.  I can switch between all 3 extruders via pronterface software.   Once I switched over to MarlinDev firmware with multi-extruder support.  It stills can extrude the primary filament via the LCD rotatory control manually.

jbrazio, 2016-04-28:
Which version of Marlin are you using ?

DIYlover, 2016-05-02:
I am using MarlinDev that downloaded from here.

thinkyhead, 2016-05-02:
Hi @DIYlover – I'm the person you want to talk to about this feature, and it definitely could use some testing. I suggest you switch away from MarlinDev and pick up the code at MarlinFirmware/Marlin#3655 instead. The full branch is at http://github.com/thinkyhead/Marlin/tree/rc_mixing_extruder.

closed 2016-05-25 by boelle