MarlinFirmware/MarlinDev #418: Interest in laser functionality in Marlin?

HakanBastedt, 2016-05-07:
I have been busy in https://github.com/MarlinFirmware/Marlin/issues/3208 where I at the end come up with an improved code for TurnkeyTyranny's laser functions.

Is there an interest to include such laser functionality in Marlin?

I have made the code changes to a copy of the MarlinDev tree except for the configuration which I am not sure how to do. Perhaps I can do a pull request for everyone to look at?

thinkyhead, 2016-05-08:
That would be great. A pull request is the way to go. Github gives a lot of documentation on that, so be sure to read it and become knowledgable first.

You should make the PR against `RCBugFix` as a starting-point, as this will have all the latest commits, and using this as a starting-point will allow you to bring in all the additional changes that we merge and keep your branch in sync with the latest work.

Keeping your branch up-to-date with the latest code-changes using `git rebase` will make it much easier to integrate when the time comes. Thus it also helps to become familiar with Git, and again there is plenty of documentation on Github and the internet generally, so you can become an expert in almost no time.

We will see about incorporating the new feature into Marlin 1.1.1 after we get the 1.1.0 release out.

emartinez167, 2016-05-08:
That is exactly what I advised in the Laserweb2 repo. I think this is a valuable addition to Marlin, so we should keep track of it. Please submit your PR so it gets applied after 1.1.0 is released. Hopefully there is not a lot of work involved in porting it to the current code base.

Regards,

Ernesto

HakanBastedt, 2016-05-08:
Ok, sounds great!
I'll prepare a pull request against RCBugFix. I can work with git on a beginner's level so I think it should be ok. 

Thanks for your suggestion @emartinez167 , I became unable to reply over there, don't know what happened. No problem to add the code to another code base.

emartinez167, 2016-05-08:
I saw the issue being closed, that might be the reason.

closed 2016-05-25 by boelle