MarlinFirmware/MarlinDev #419: moving issues?

boelle, 2016-05-08:
@thinkyhead Scott, is it possible to move issues across repro's?

jbrazio, 2016-05-08:
no

jbrazio, 2016-05-08:
(only manually or API)

boelle, 2016-05-08:
so with API it becomes a yes anyway

CONSULitAS, 2016-05-08:
@boelle @thinkyhead 
Perhaps Google is our friend here:
https://www.google.de/search?q=github+move+issues+between+repositories&ie=UTF-8&oe=UTF-8&hl=de-de

https://github.com/google/github-issue-mover

boelle, 2016-05-08:
yep, i used that one before on some other gihub work...

as long you have write access at both ends it works

thinkyhead, 2016-05-09:
The auto-mover doesn't get everything. For the moment I vote we just keep both repos open until we get this release out. Some good discussions are still open around here (for better or worse).

Roxy-3DPrintBoard, 2016-05-09:
> For the moment I vote we just keep both repos open until we get this release out. Some good discussions are still open around here (for better or worse).

When we know we won't lose anything...   (When the time is right)   My vote is it can go away...

If we are going to start a new thread for this:  https://github.com/MarlinFirmware/MarlinDev/issues/398

I guess it should be done on the Marlin side of the fence.

jbrazio, 2016-05-09:
I backup @thinkyhead.

closed 2016-05-25 by boelle