MarlinFirmware/MarlinDev #421: Unable to use M421 commands due to usage conflict

bdowling, 2016-05-12:
M421 are unable to be set due to a conflict in the code regarding what X and Y values are to be passed.  Well, technically, I can set the X0, Y0 ;)

https://github.com/MarlinFirmware/MarlinDev/blob/806d8861818a80b5875f0570f9212ea88f5cae67/Marlin_main.cpp#L5198

The test of the parameters are checking that an index is being passed, but then those values are being handed off to the mapping functions mbl.select_x_index(x) and respective y  rather than being used as an index.

I would like to propose that the interface be modified such that the output from M503 can be pasted back into the serial to be able to set/reload these parameters if so needed.  Currently M503 outputs the absolute X and Y coordinates.  It could go either way, change the M503, or change M421 to accept the physical coordinates.  I'm not sure how widely this is used.  I think I must be the first to try. ;)

Ref: http://reprap.org/wiki/G-code#M421:_Set_a_Mesh_Bed_Leveling_Z_coordinate

Willing to submit a PR on this, just wanted to open the discussion for the above.

bdowling, 2016-05-12:
Example test case:

M421 X0 Y0 Z1
M421 X1 Y0 Z2
M421 X2 Y0 Z3
M421 X0 Y1 Z4
M421 X1 Y1 Z5
M421 X2 Y1 Z6
M421 X0 Y2 Z7
M421 X1 Y2 Z8
M421 X2 Y2 Z9

M503   
; Relevant section
Recv: Mesh bed leveling:
Recv: echo:  M420 S1 X3 Y3
Recv: echo:  M421 X5.00 Y5.00 Z9.00
Recv: echo:  M421 X130.00 Y5.00 Z2.55
Recv: echo:  M421 X255.00 Y5.00 Z2.25
Recv: echo:  M421 X5.00 Y100.00 Z2.83
Recv: echo:  M421 X130.00 Y100.00 Z2.88
Recv: echo:  M421 X255.00 Y100.00 Z2.58
Recv: echo:  M421 X5.00 Y195.00 Z2.50
Recv: echo:  M421 X130.00 Y195.00 Z2.35
Recv: echo:  M421 X255.00 Y195.00 Z2.20

lrpirlet, 2016-05-13:
@bdowling Well, my answer to that was to modify the power on report when it gets the values from the EEPROM... see configuration_store.cpp line 698-706 in my *Marlin-motorized_mesh_bed_leveling_for_LRP* branch https://github.com/lrpirlet/Marlin.git 
```cpp
    for (int y = 0; y < MESH_NUM_Y_POINTS; y++) {
      for (int x = 0; x < MESH_NUM_X_POINTS; x++) {
        CONFIG_ECHO_START;
        SERIAL_ECHOPAIR("  M421 index X", x);
        SERIAL_ECHOPAIR("Y", y);
        SERIAL_ECHOPAIR(" X", mbl.get_x(x));
        SERIAL_ECHOPAIR(" Y", mbl.get_y(y));
        SERIAL_ECHOPAIR(" Z", mbl.z_values[y][x]);
        SERIAL_EOL;
      }
    }
  #endif
```
To be honest, after I wrote it, I forgot why I needed it... and never tested the M421... Was it because MBL was motorized and the need to correct a single point just fade away?
Again, I am not going to support it, but it is open software and you may get an idea out of my work...
@epatel thanks for your work that I used to produce "my" motorized MBL...

bdowling, 2016-05-14:
I'm going to close this issue one since it duplicates the ticket in MalinFirmware/Marlin.

closed 2016-05-14 by bdowling

gddeen, 2016-05-15:
M421 didn't change did it???
I use G29 S4 Z0.3 I assume it is the same thing but for the home point 120, 120 (or so...)