MarlinFirmware/MarlinDev #422: XY angle compensation

MLWORKX, 2016-05-13:
This is a re-opening as it was closed on an /Marlin
MarlinFirmware/Marlin#3014

_"Hi,
even when the (cartesian) printer is calibrated very well there is no way of calibration the correct 90° angle of the XY plane: you have to do this on the hardware side via loosening the belt gears and hope you did it correctly.
So like the bed compensation where the Z axis moves when X and Y are moving it would be a nice feature to have a XY calibration possibility as e.g. the M92 feature for the dimensioning.
Just add a compensation angle (e.g. 0,2 degrees) when the angle between X and Y is not exactly 90° and no hassle with the belt anymore."_

Roxy-3DPrintBoard, 2016-05-14:
This can be done and would be fairly straight forward because the Auto Bed Leveling (Grid or 3-Point) maps a specified coordinate to a new location.   But getting the X & Y square is not difficult on most machines.   I'm struggling to see a benefit when the machine can just be adjusted to have X & Y normal to each other.   And assuming the case can be made, it does add complexity for people that don't need this (which is the bulk of the users).

bdowling, 2016-05-15:
Wouldn't it also likely result in non smooth veritcle sides of xy planar surfaces with all that micro stepping?

This seems like a physical fix is the better solution.  

Is this request specific to gantry style machines?

On a rod/rail style printer your likely stressing the bearings and bit and not smooth as it could be.

MLWORKX, 2016-05-15:
> I'm struggling to see a benefit when the machine can just be adjusted to have X & Y normal to each other. 

The thing is that there are several printers on the market where the rails, the belts and everything which slines linear tends to be in a very tiny space: so it's hard to adjust and to unlock a screw, push the rod in a direction and re-adjust the screw will NEVER be as accurate on printers which large build volume

> And assuming the case can be made, it does add complexity for people that don't need this 

Well it does, you're right. But adding a short comment in configuration.h to explain and set this compensation value to 0 would not add complexity to the system I guess - not more than all those PID, autolevel,... features

> This seems like a physical fix is the better solution.

Yes it is: but there are machines where this is absolutely impossible as there is no space to adjust

> Is this request specific to gantry style machines?

It's an xBot 155 printer and I'm struggling since weeks to print an accurate 90° angle.

From my point of view it would make machines much more precise if you just could add something in configuration.h (or do it via an M command) and I also do not think that it would make the system more complex. Those who want to fine-tune their machines -and mine do far better than every >30k Stratasys machine- it would make their parts a bit more accurate and perfect.

closed 2016-05-25 by boelle